RoboJackets/igvc-software #43: Set up formatting enforcement in our make process

thaeds, 2015-09-20:
Robocup is using the following python pretty script in their makefile (https://github.com/RoboJackets/robocup-software/blob/master/makefile) to enforce their formatting guidelines during build.

```
# automatically format code according to our style config defined in .clang-format
pretty:
	@stylize --diffbase=master --clang_style=file --yapf_style=file --exclude_dirs $(STYLE_EXCLUDE_DIRS)
```

This references the .clang-format style-configs here https://github.com/RoboJackets/style-configs/blob/master/.clang-format

We want to use this same format file, so we enforce the same styleguide when making that robocup does.

Since we use catkin_make, we need to figure out where to execute this script. Possible in CMakeLists.txt somewhere?

Please ask @justbuchanan for help as needed!

rmkeezer, 2017-02-26:
Since we are using catkin cmake instead of make, the build system is a bit different. For now, we can use clang-format to format any one file or a bash file to format all files.

jgkamat, 2017-02-27:
If you want, you can use justin's tool called stylize to format only the files changed since the merge from master (which is what we use). You can write a small shell script to format all files if you want that intead as well.

barulicm, 2017-02-27:
Our ROS projects should follow the ROS style guidelines to be consistent with other ROS code out there.
There is a clang-format style definition for the ROS guidelines available [here](https://github.com/davetcoleman/roscpp_code_format).

rmkeezer, 2017-03-02:
I was using the robojackets formatting that Robocup are using: https://github.com/RoboJackets/style-configs
I'm not a fan of using ros style guide since they are quite a bit different from what our current code is.
2 space indents
braces on their own line
120 column limit
But if everyone is ok with using the ROS style guidelines, we'll use that.

rmkeezer, 2017-03-02:
I'll open up a poll

rmkeezer, 2017-03-09:
ROS style is being used for now but may be changed in the future. Auto-formatting is moved and detailed in issue #192

closed 2017-03-09 by rmkeezer