RoboJackets/igvc-software #68: Reduce GPS Noise through Kalman Filter

thaeds, 2015-11-30:
Another type of filter is acceptable if more appropriate, but GPS noise must be reduced through sensor fusion, possibly through using a Kalman Filter. The problem is observable as either the robot believing it is jumping around its physical location while it is actually stationary (noise), or the robot believing it is moving in a certain direction while it is actually stationary (drift).

Implementing sensor fusion through a Kalman Filter will allow us to estimate our current position with more accuracy and less noise, given that we will be able to take into account our current wheel odometer y.

AlexTrimm, 2015-11-30:
There should be code sitting around somewhere for this. We attempted it 2-3
years ago, but we were never able to tune it well enough to make it useful.
Matt and Al should be passingly familiar with it. I'd suggest talking to a
controls professor about effective methods for tuning the parameters. Dr.
Feron out of the AE school has been involved in student competition teams
in the past. He might be worth asking if you don't know of someone
personally. If you're looking to learn more about Kalman Filters, the
original paper is quite good.

thaeds, 2016-01-28:
That sounds awesome - I think the first place to look is the ROS implementation
http://wiki.ros.org/robot_pose_ekf

But @barulicm has noted that in the past this implementation has not been good enough to use.

Certainly reading the original paper would be awesome regardless. (below)
https://www.cs.unc.edu/~welch/kalman/media/pdf/Kalman1960.pdf

barulicm, 2016-09-19:
Since this issue is coming back, I'll note that it's worth looking into [robot_pose_ekf](http://wiki.ros.org/robot_pose_ekf) again, as a significant amount of development has been put into it since I last looked at it.

jgkamat, 2016-12-01:
@rmkeezer I think this issue can be closed? Referenced in #144

dpattison3, 2016-12-01:
It was a little hasty to say that it solved that issue. It addresses the issue, but it still needs to be tested in competition scenarios, so I think it should remain open until I can show it working well in those conditions.

dpattison3, 2018-01-07:
The localization node has been tested, so I think we can close this now

closed 2018-01-07 by dpattison3