RoboJackets/igvc-software #73: Adding 2 new cameras

thaeds, 2015-11-30:
Two new downwards facing cameras are to be added to the front of the robot for higher fidelity detection of obstacles in the immediate vicinity.

These two new cameras need calibration, to be included in the tf tree, to have overhead transforms, and to publish the resulting images to sensible topics.

The current architecture for the BumbleBee can be copied somewhat, only simplified, as it is a stereo camera.

dpattison3, 2016-09-17:
I believe this has been accomplished. Or at least the issue should be narrowed significantly to specify what still needs being done.

jzheng84, 2016-09-18:
I believe the cameras have been localized, and we were able to map lines detected from those cameras into the point cloud. Couple things that still have to be done:
1) Improve localization (currently the cameras do not look like they are in the correct position on the tf tree)
2) Crop out any unnecessary parts of the image (i.e the vehicle)
3) Find an ideal angle for the cameras to be mounted onto the vehicle so we have a large enough field of view

kscharm, 2016-10-19:
We have ordered two new side cameras for the new robot. I will be working on the three points mentioned by @jzheng84.

rmkeezer, 2017-03-09:
The new cameras are already in the TF tree and work in Gazebo. This is completed but more testing should be done when the robot is complete.

kscharm, 2017-08-27:
We want a tool that can do this for any set up of the cameras. It is possible that the cameras could shift over time. It may be good for setting up new cameras or changing their positions.

atuttle7, 2018-11-01:
No longer applicable to current platform.

closed 2018-11-01 by atuttle7