RoboJackets/igvc-software #82: Test path planner performance under competition scenario

thaeds, 2016-01-28:
Look into the competition rules and determine how far away it is possible for a waypoint to be in the worst case. Run the path planner under the following scenarios:

No possible path
Obstacle which can be gone around
Obstacles which create a "divet" the robot could potentially get stuck in

Record the following:
Number of nodes created during path planning
Time taken to plan the path
Timestep the planner is taking during search
Branching factor (and start/stop angles)

You will likely need to come to the test track and make sure we bag data of the robot in the required situations to have a good, easy testing environment.

barulicm, 2016-01-28:
@jgkamat Did the CI stuff include a testing setup? If we have one, we should make some of these permanent tests that get checked regularly.

thaeds, 2016-01-28:
Ooo that would be awesome

jgkamat, 2016-01-28:
Yup, just let me know the command you use to run the tests, and I can add them. They will show up as additional checkmarks in the pull requests and stuff.

Alescontrela, 2019-02-09:
Path planner updated.

closed 2019-02-09 by Alescontrela