RoboJackets/igvc-software #83: Setup ROS-aware testing framework

barulicm, 2016-01-29:
Make necessary changes to enable the use of rostest and gtest in our code base to test the functionality of libraries, nodes, and subsystems.

barulicm, 2016-01-30:
@jgkamat How does the CI system interpret the test output for the fancy "test passed" checks?

While ROS's testing system uses gtest under the hood, it hides the default gtest output, instead directing the output to a special xml file. The file below contains an example of the output the test command actually spits out.

[output.txt](https://github.com/RoboJackets/igvc-software/files/110562/output.txt)

barulicm, 2016-01-30:
@jgkamat the test command for the CI to run is `catkin_make run_tests`

jgkamat, 2016-01-30:
To add this you would add a line like this [in this file](https://github.com/RoboJackets/igvc-software/blob/master/config.docif#L94)

`TEST_COMMANDS+=( 'catkin_make run_tests;tests;A check to see if the tests pass' )`

By default, the output will be captured and uploaded to the details link of this check. At this point, the easiest thing to do would be printing the output.xml file to stdout and letting the defaults take care of that.

jgkamat, 2016-01-30:
If you want, I can add this/test it, should I make a seperate PR for that?

JasonGibson274, 2017-01-13:
The testing framework is in place just no tests have been written.

closed 2017-01-13 by JasonGibson274