RoboJackets/igvc-software #110: Global Mapping

rmkeezer, 2016-09-14:
We are currently performing path-finding using point clouds whose positions are in relation to the robot and in order to create these point clouds we are converting GPS latitude and longitude data to local xyz map where our robot is the origin. We should improve our system by mapping in relation to some global position.

jzheng84, 2017-04-02:
Currently the line detector and pothole publish two orders of magnitude too many points. This causes ICP and the path planner to slow down as more points are added to the global map.

Potential fixes:
1) Reduce image size
2) Use interpolation in the line detector to plot sparser lines 

To reduce the computation time for ICP and the path planner do not add points if the distance to their nearest neighbor is less than some epsilon.

Note: Points can have the same coordinates

jzheng84, 2017-05-04:
Number of points added has been significantly reduced, now needs to be tested with functional line detector with barrels.

jzheng84, 2017-11-30:
#241

closed 2017-11-30 by jzheng84