RoboJackets/igvc-software #125: Stop throwing char* or any other primitives/pointers/etc

red-eight, 2016-10-20:
We should throw C++ exception classes instead of primitives or pointers.

The [<stdexcept> header](http://en.cppreference.com/w/cpp/header/stdexcept) has about 10 different type of exception classes; all of them inherit from [std::exception](http://en.cppreference.com/w/cpp/error/exception). 

Because all of these classes inherit from `std::exception`, we only have to write: `catch (const std::exception &e)`, but we can throw any of these types of exceptions. We can also create our own exceptions, but they should also inherit from `std::exception`.

rmkeezer, 2016-10-23:
Good idea! This will also help with creating working test cases (when someone decides to add some)

sunny3728, 2017-02-12:
Already been fixed.

closed 2017-02-12 by sunny3728