RoboJackets/igvc-software #149: Investigate possible race condition in building igvc_msgs

rmkeezer, 2016-12-05:
CircleCI is sometimes not building igvc_msgs before igvc. This should have been fixed a while ago. If anyone is failing builds on master please post the log here

dpattison3, 2016-12-05:
Rviz plugings are the only module that doesn't set messages as a dependency (it wasn't compiling when I fixed the earlier issue anyway). Should it depend on messages? That's the only recent thing to change right?

JasonGibson274, 2017-01-13:
I think we should close this issue and reopen it if the issue returns. CircleCI seems to be building fine to me.

jgkamat, 2017-02-13:
This is still happening, just at a much reduced rate. [See this build log for details](https://271-16996386-gh.circle-artifacts.com/0/tmp/circle-artifacts.rnZw4Il/compile.txt). I /think/ that rviz_plugins needs to depend on igvc_msgs as well (although, since igvc depends on it, shouldn't igvc_msgs depend on it as well?).

commit ae4b00f1 (2017-02-13): Try to fix igvc_msgs race condition again

Closes #149
commit f342eced (2017-02-13): Try to fix igvc_msgs race condition again

Closes #149
PR #173 (2017-02-16): Try to fix igvc_msgs race condition again
Closes #149 

Let me know if this dosen't seem right. No way to be sure if this is fixed without letting it sit for a long time.
closed 2017-02-16 by rmkeezer