RoboJackets/igvc-software #216: Update Readme

JasonGibson274, 2017-06-01:
This will be complete when the Readme is accurate and more specific documentation has been moved to the wiki and linked in the Readme.

JasonGibson274, 2017-09-02:
make format is not in the readme. Also look at ways of doing this through catkin automatically when catkin_make is run. Also the node list at the bottom is a couple years old.

jgkamat, 2017-09-02:
Also, it would be nice to add a picture of the robot and/or a picture of igvc-software running, just to give people something to expect :).

JasonGibson274, 2017-09-02:
add launch instructions for both the physical robot and in gazebo

JasonGibson274, 2017-09-03:
Make a separate markdown file for the nodes and their published and subscribed topics.

JasonGibson274, 2017-09-03:
add command to run tests to this readme

atuttle7, 2018-11-01:
Readme should include build concerns and instructions

Alescontrela, 2019-02-11:
Still relevant. Detailed wiki update should take place after new planner/occupancy grid are merged into devel.

oswinso, 2019-07-06:
Closing in favor of #454.

closed 2019-07-06 by oswinso