RoboJackets/igvc-software #225: Create a node for recording the speed of the robot over time

dpattison3, 2017-08-27:
This will be useful as a step in validating the PID tuning. We want to be able to compare the speed published by /motors and the speed recorded on /encoders. This can be done with just a csv file, then plotted in excel.

dpattison3, 2017-08-31:
related to #224

AaronMcDaniel, 2018-02-15:
Yo, lemme do this!

AaronMcDaniel, 2018-02-15:
Never mind. I can assign myself now.

barulicm, 2018-02-15:
Is this intended to have any functionality beyond what can be accomplished with `rosbag record` and `rqt_plot` (or `rqt_bag`'s plot feature)?

JasonGibson274, 2018-02-16:
I think a more quantitative measurement of PID tuning would be interesting. I also think that it is a good node to start working with in order to understand how PID works.

JasonGibson274, 2018-02-16:
Also rqt_plot does not work super well without timestamps which I think this issue will also include resolving parts of #286

atuttle7, 2018-11-01:
Done. :)

closed 2018-11-01 by atuttle7