RoboJackets/igvc-software #232: update path follower to use closed loop control

dpattison3, 2017-08-31:
The path follower is currently open loop. We would like for this to subscribe to the encoders or position to better follow paths.

JasonGibson274, 2017-08-31:
I 100% believe that this should be implemented. I do not see how this is a freshmen project at all though? What exactly do you want the freshmen to do here?

Nam-Nor, 2017-11-06:
I'm really interested in doing this since I have a bit of control system design experience. Would you be able to provide more details regarding this @JasonGibson274?

JasonGibson274, 2017-11-06:
Currently our path planner is completely open loop control. We plan entirely in kinematic space and our path planner directly outputs motors commands. The path follower does nothing more than just sending those motor commands down to the arduino which controls our motors. We would like to be able to have the path follower node be able to drive to a point in the global reference frame without the path planner giving it explicit directions.

JasonGibson274, 2017-11-06:
You do not have to worry about altering the path planner or obstacle avoidance just getting from the current location to a given point.

Alescontrela, 2019-02-09:
Still relevant. Not implemented in codebase, therefore not a bug. Currently being researched by @JasonGibson274

closed 2019-02-09 by Alescontrela