RoboJackets/igvc-software #241: Probabilistic mapping

jzheng84, 2017-09-24:
The planner should be robust to erroneous points included in the point cloud from our sensors and mapping algorithm. The mapper should filter out these points based on the amount a point is added to a point cloud over the last x number of frames. If this is implemented correctly, false positives will be less of an issue for the line detector and pothole detector.

dpattison3, 2018-01-07:
I think we should move towards a true SLAM approach if we have time. There are a few libraries to consider:
https://github.com/OctoMap/octomap
https://github.com/ethz-asl/grid_map
https://github.com/CognitiveRobotics/omnimapper

atuttle7, 2018-11-01:
Should include a decay of some sort

Alescontrela, 2019-02-09:
Completed by Oswin. Still not merged into development branch, therefore not a bug

closed 2019-02-09 by Alescontrela