RoboJackets/igvc-software #247: Convert image processing nodes to nodelets

JasonGibson274, 2017-10-19:
copying of the image so many times is inefficient. We need to convert out image processing nodes to nodelets so that is can be done faster.

atuttle7, 2018-11-01:
1.) Convert python TF node to C++
2.) Convert to nodlet

Alescontrela, 2019-02-11:
Perhaps a better alternative is to convert image processing nodes into [ROS Services](http://wiki.ros.org/Services), which are for all intents and purposes non-blocking processes.

JasonGibson274, 2019-02-11:
That is not why this exists at all. It is to save time by passing image as a pointer rather than through network stack. camera driver is not very intensive. Cannot do until we move to cpp on vision stuff though

Alescontrela, 2019-02-17:
Ok, i'll backlog this

closed 2019-02-17 by Alescontrela