RoboJackets/igvc-software #264: Train a CNN on line segmentation data

rmkeezer, 2017-11-26:
I am currently trying to modify KittiSeg which is a fully convolutional neural network (FCN or CNN) written in tensorflow that outputs tensorflow models.

https://github.com/MarvinTeichmann/KittiSeg
https://www.tensorflow.org/install/

I recommend using these from my exploration on CNN tools. Other options include using Caffe2, Theano, or PyTorch and some other semantic segmentation repo on github.

dpattison3, 2018-01-07:
KittiSeg seems to work well. We now need to make it run in a ros node on the robot

rmkeezer, 2018-01-10:
Update: the tensorflow kittiseg model does run in a rosnode using python. The performance might be better if we are able to load the model in C++

dpattison3, 2018-03-25:
Addressed by pull request #295

closed 2018-03-25 by dpattison3