RoboJackets/igvc-software #265: Update OpenCV to include version 3.3 in order to use DNN module

rmkeezer, 2017-11-26:
Our current OpenCV version is 3.2 which is installed using rosdep which uses a custom apt archive that installs the newest version of OpenCV.

According to the ros repositories, opencv has recently been updated to 3.3 but it does not appear to have been updated to rosdep yet.

http://repositories.ros.org/status_page/compare_indigo_jade_kinetic_lunar.html?q=opencv3

If we can get the DNN module from opencv (which is only available in 3.3 without compiling) we can avoid compiling opencv from source and will be able to read ML modules from tensorflow, theano, etc.

kscharm, 2017-11-30:
@rmkeezer What actually needs to be changed in the code to support version 3.3? How do we ensure that the latest version is installed?

rmkeezer, 2018-01-10:
We currently can load and run the tensorflow model in python and subscribing and publishing the results using ROS. However, this might be much slower than using C++ opencv.

dpattison3, 2018-01-22:
I'm not sure on the details, but I think that if everything is up to date, then ros does use opencv 3.3. At the very least I have imported the dnn module to a node (as I have opencv 3.3 installed).
They have a convenient function readNetFromTensorflow. This has some limitations though and our current model won't load as our graph uses unsupported layer types (namely expand dimensions). I think it might be worthwhile to investigate how to remove these layers (and others that might be unused).

dpattison3, 2018-03-25:
I don't think this is the best way to do this. Either we should go directly with tensorflow's c/c++ api, or use tensorrt.

closed 2018-03-25 by dpattison3