RoboJackets/igvc-software #288: handle lidar bias

JasonGibson274, 2018-02-08:
Our lidar has intrinsic bias due to it not being correctly mounted and it results in smearing of the global map since the robot thinks is has a constant acceleration on the y. We need a way to handle this bias in software so that we are not dependent on the imu being mounted perfectly flat. A logical place to start would be reading in the values over a long time and then canceling out the average vector.

barulicm, 2018-02-08:
I think you've substituted 'lidar' for 'imu' in a few places in this issue.

JasonGibson274, 2018-02-08:
Fixed that issue

Alescontrela, 2019-03-07:
Still relevant? I have not witnessed this behavior with the new imus. The only remaining Imu bias is that of electromagnetic interference with the magnetometer.

JasonGibson274, 2019-03-07:
That’s what this is about

closed 2019-07-06 by oswinso