RoboJackets/igvc-software #326: ROS Melodic / Ubuntu 18.04 Readiness

barulicm, 2018-07-19:
Congratulations! I did not find any issues **building** your code with Melodic / 18.04.

However, I did notice that igvc/launch/camera.launch depends on usb_cam. This dependency is not noted in igvc's package.xml, so rosdep won't try to install it. Note that usb_cam is not yet released in Melodic and will have to be built from source.

Please double check your dependency specifications. There may be other run dependencies in the project that have not yet been released in Melodic.

barulicm, 2018-07-24:
usb_cam is expected to be released as part of the next Melodic sync this Friday. You should not have to build it from source when you switch.

Alescontrela, 2019-02-09:
Still relevant?

barulicm, 2019-02-10:
Given you have all been running Melodic/18.04 for the entire school year so far, probably not.
This was from last summer as part of a multi-project audit of Melodic readiness.

Alescontrela, 2019-02-11:
I figured. Closing this then.

closed 2019-02-11 by Alescontrela