RoboJackets/igvc-software #329: Add warnings to path planner

JasonGibson274, 2018-08-26:
The path planner should warn if the distance to the goal is too far away

Alescontrela, 2018-09-27:
viaa rviz, the rqt console, or the terminal?

JasonGibson274, 2018-09-27:
just the terminal

Alescontrela, 2018-10-03:
So at [this line](https://github.com/RoboJackets/igvc-software/blob/716341fb930ff60d77413d953920ad2fd5ef3b83/igvc/src/pathplanner/main.cpp#L154) in pathplanner/main.cpp, a check is made to ensure that the distance to a goal is not greater than a hard-coded value (60m, in this case).

One possible option would be to move this presently hard-coded value into the parameter server. This value can then be accessed using the NodeHandle defined [here](https://github.com/RoboJackets/igvc-software/blob/716341fb930ff60d77413d953920ad2fd5ef3b83/igvc/src/pathplanner/main.cpp#L108). 

We could also define a threshold value which can be compared against `distance_to_goal` to decide whether or not to print output messages to the terminal. If the max distance value is 60m and the threshold value is 40m, then a `distance_to_goal` value of 45m would create a warning in the terminal but the search would still be performed.

Alescontrela, 2019-02-09:
This is implemented with new path planners.

closed 2019-02-09 by Alescontrela