RoboJackets/igvc-software #394: Convert ground_truth to be a class and not use global variables

oswinso, 2019-04-18:
Globals are currently used in `ground_truth.cpp`, and can be converted into a class in order to avoid them. Ex:

https://github.com/RoboJackets/igvc-software/blob/5f1aadc51b36499a7dd5f7147ea61ff6a8763878/igvc_gazebo/nodes/ground_truth/main.cpp#L11-L14

> Global variables should almost never be used (see below for more on this). When they are used, global variables are under_scored with a leading g_ added. 

http://wiki.ros.org/CppStyleGuide#Global_variables

JasonGibson274, 2019-04-18:
Maybe we should create an issue per node that needs this.

oswinso, 2019-04-18:
Yeah, I agree. My plan kind of was to create these issues in one of the sessions next year when we do like a code review of some code, so that new members get an example of what it's like to create issues for problems they see in the code base + they can get the opportunity to work on an issue of a class which was just gone over during training.

matthewhannay567, 2020-09-30:
This is now #739

closed 2020-09-30 by matthewhannay567