RoboJackets/igvc-software #571: Turn the cone of shame into the cone of fame

oswinso, 2019-10-31:
To prevent going backwards, we need to legitimize the cone of shame, and properly integrate it into the mapper.

AC: Jessii doesn

tangemicioglu, 2019-11-04:
Based on behavior I've seen in autonav, I think we'll need the cone of fame at every waypoint.

JasonGibson274, 2019-11-04:
Really? That seems wrong, it should only really be required for the first way point I think. Can you attach a picture with waypoints, course, and proposed global path?

oswinso, 2019-11-04:
Yeah, I agree with Jason. For the second to third, the straight line should be the correct global bath. For the third to the fourth, because we already have the global map it's going to plan correctly.

tangemicioglu, 2019-11-04:
Second to third is the problematic one, the only "straight line" you get there is when the robot goes outside of the lines. I can't pull up the sim right now, but here's a drawing of what happens:
[image]

oswinso, 2019-11-04:
oh, interesting. I guess that makes sense. Yeah, we'll probably need to do something to make sure that that's not valid.

JasonGibson274, 2019-11-04:
Do we have to go across the bridge? That looks like a valid path?

oswinso, 2019-11-04:
I think we need to go through the center thing. I haven't checked the rulebook for a specific rule, but I get the feeling that we aren't supposed to go that way.

JasonGibson274, 2019-11-04:
I don't see that we have to over the ramp in the rules. Yes it is faster, but it is not required. You just need to get to the waypoints.

tangemicioglu, 2019-11-04:
That feels a bit cheaty especially now that they've removed the ramp and replaced it with a barrel course, but I can't find any rules forbidding us now that I look around the rulebook. Is there a way for us to ask the organizers about this scenario?

JasonGibson274, 2019-11-04:
We can ask them, but the whole points is whatever legal path is fine. So avoiding it is reasonable in the legality sense. The questions is are we okay with taking the somewhat longer path as a software team?

tangemicioglu, 2019-11-04:
It's actually a shorter path because the obstacles in that area make the path between the lines longer in the 2020 layout, so I think it's perfectly fine as long as it's legal.

oswinso, 2019-11-04:
>  One waypoint pair will be the entrance and exit of the course in No Man’s Land

> Each vehicle must navigate the course by remaining inside the course boundaries

So I think because they define it as a "course", then we have to stay inside

JasonGibson274, 2019-11-05:
Inside the outer boundary, or inside the arbitrary lines of the ramp? because at that point why does No Man's Land exist at all??

tangemicioglu, 2019-11-05:
Without No Man's Land, you can essentially ignore waypoints and make a lane following robot. By disjointing a portion of the course from the main body, it makes sure that we actually care about waypoints and are able to navigate to one.

JasonGibson274, 2019-11-05:
So the conclusion is that we do not have to go over the ramp and the cone of shame is only needed okay the first waypoint?

oswinso, 2019-11-05:
The ramp is no longer there anymore btw. It's a "separate" course, but it still is a valid course so we need to stay between the white lines and actually go through all the barrels inside I think.

JasonGibson274, 2019-11-05:
Wat? Why do we have to go through that space rather than just going around it. Both should be valid (not hit obstacles or cross white lines) or am I missing a large point here?

oswinso, 2019-11-05:
I think the behavior that the competition wants is "lane following", and if you just ignore the white lines in front of you and go around it I don't think that counts as "lane following". Whether we can or cannot go around it though I think it would be worth to implement something that's toggleable, so that during the competition if we get clarifications or whatever we can always just add / remove it.

matthewhannay567, 2020-09-30:
Moved to #749 and #750

closed 2020-09-30 by matthewhannay567