RoboJackets/igvc-software #575: Joystick tests nondeterministically failing

oswinso, 2019-11-01:
So... for some reason b50471d resulted in joystick tests failing.... Searching "joystick test", I found #130, so apparently we've had the same nondeterministic joystick test for like 3 years. 99% this is a race condition somewhere and we're all dumb.

Some instances of it happening:
- b50471d
- 1e28df70d7ff1f77c8376c5b3a6ed1832fb209a6

AC: (Hopefully) fixed joystick tests from failing non-deterministically

oswinso, 2019-11-01:
Okay of-fucking course this is a race condition. Why am I so dumb.

https://github.com/RoboJackets/igvc-software/blob/8525035aabf1a017ee630bb4b166e8086d0a7b87/igvc_platform/src/tests/test_joystick_driver.cpp#L49-L55

Here, we publish a message first, and then call `waitForMessage`. I guess we were all thinking that `waitForMessage was like a magic function, but here's the [implementation](https://github.com/strawlab/ros_comm/blob/6f7ea2feeb3c890699518cb6eb3d33faa15c5306/clients/cpp/roscpp/src/libros/topic.cpp#L43-L62):

```c++
void waitForMessageImpl(SubscribeOptions& ops, 
			const boost::function<bool(void)>& ready_pred, 
			NodeHandle& nh, ros::Duration timeout)
{
  ros::CallbackQueue queue;
  ops.callback_queue = &queue;

  ros::Subscriber sub = nh.subscribe(ops);

  ros::Time end = ros::Time::now() + timeout;
  while (!ready_pred() && nh.ok())
  {
    queue.callAvailable(ros::WallDuration(0.1));

    if (!timeout.isZero() && ros::Time::now() >= end)
    {
      return;
    }
  }
}
```

So, it actually **creates a new subscriber**. Except, of course, we published the message first before creating the subscriber that we expected to be able to somehow catch that message we just published.

Now, you might ask, "How come this works most of the time?"

Well, let's look at how `publish()` actually works:
There's 3 layers of `publish()`, each calling the next:
1. [Publisher](https://github.com/ros/ros_comm/blob/5da095d06bccbea708394b399215d8a066797266/clients/roscpp/src/libros/publisher.cpp#L78-L101)
2. [TopicManager](https://github.com/ros/ros_comm/blob/5da095d06bccbea708394b399215d8a066797266/clients/roscpp/src/libros/topic_manager.cpp#L704-L760)
3. [Publication](https://github.com/ros/ros_comm/blob/5da095d06bccbea708394b399215d8a066797266/clients/roscpp/src/libros/publication.cpp#L407-L431)

Now, `Publication::publish` adds the message to `publish_queue_`.

This `publish_queue_` is processed in `Publication::processPublishQueue`, which calls `Publication::enqueueMessage`, which iterates over all the subscribers and calls the subscriber's `enqueueMessage`, which I won't go further into.

One important thing is that `Publication::processPublishQueue` is called by `Publication::processPublishQueue`, which in turn is called when a `boost::signals2::signal poll_signal_` is signalled (think of QT slots and signals). Now, when is this `poll_signal_` signal signalled? 

It turns out there's a `threadFunc` in `PollManager` that calls it in a while loop that runs forever.

```c++
void PollManager::threadFunc()
{
  while (!shutting_down_)
  {
      poll_signal_();
  }
}
```

Doesn't this while loop look really suspicious? What calls `PollManager::threadFunc()`? Well, we just **spin up a thread** for this:
```c++
thread_ = boost::thread(&PollManager::threadFunc, this);
```
Ok. And then it turns out we spin this new thread when `ros::start()` is called, which is called when we construct a `ros::NodeHandle`.

Okay, that was a very long detour resulting in finding a separate thread that is spun up. Speaking of new threads that are spun up:

https://github.com/RoboJackets/igvc-software/blob/8525035aabf1a017ee630bb4b166e8086d0a7b87/igvc_platform/src/tests/test_joystick_driver.cpp#L127-L128

So, we've got _another_ thread, from `ros::AsyncSpinner`. 

Let's recap what we have right now:
[... 10 lines cut ...]
We need things to happen in the order 1, 2, 3. If things happen in 1, 3, 2 (Which could happen because 2 and 3 are on separate threads), then `publish_queue_` is sent off to all subscribers, but this is **before** the `waitForMessage` Subscriber is added to the list of subscribers. This means that **waitForMessage** will be waiting for a message that no longer exists.

This means that we have a race condition between 2, and 3, which results in the nondeterministic nature of this bug. Most of the time, the order is 1, 2, 3. But occasionally, we get 1, 3, 2, and stuff stops working.

There's still one unanswered question though: Why is the error
```
TestJoystickDriver: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = const igvc_msgs::velocity_pair_<std::allocator<void> >; typename boost::detail::sp_member_access<T>::type = const igvc_msgs::velocity_pair_<std::allocator<void> >*]: Assertion `px != 0' failed.
```
ie. accessing a null `boost::shared_ptr`, when we'd expect it to error out on the null check:
https://github.com/RoboJackets/igvc-software/blob/8525035aabf1a017ee630bb4b166e8086d0a7b87/igvc_platform/src/tests/test_joystick_driver.cpp#L54-L59

The answer to this is embarrassingly simple: IT'S A FUCKING EXPECT AND NOT ASSERT. From [gtest](https://github.com/google/googletest/blob/master/googletest/docs/primer.md):
> The assertions come in pairs that test the same thing but have different effects on the current function. ASSERT_* versions generate fatal failures when they fail, and abort the current function. EXPECT_* versions generate nonfatal failures, which don't abort the current function.

So the expect generates a nonfatal failure, and then executions continues to the next line where we dereference the `nullptr`. Checkmate.

So, the full story (chronological) of how this crash is happening
1. We publish a message and add it to the `publish_queue_`
2. The separate thread empties the `publish_queue_`
3. We call `waitForMessage`, except the `publish_queue_` is already emptied
[... cut ...] rns a `nullptr`
5. The call to `EXPECT_TRUE(response.get() != nullptr)` fails, but since it's an `EXPECT` execution continues
6. In the next line, `EXPECT_EQ(response->left_velocity, 1.0);`, we dereference `response` which is empty, generating the crash.

To avoid this, we need to avoid the anti-pattern of calling `publish()` and then doing `waitForMessage` right after. I've got a "test framework" that solves this problem I wrote up this summer, so I think I'll just copy-pasta that for use here.

barulicm, 2019-11-01:
Nice work diving into this. This test was written before I'd learned all I know now about the ROS 1 backend, so I'm not super surprised this mistake happened. 

There is basically zero documentation on how you should actually write a node-level test like this. I think one really good thing we could do is figure out what the patterns should be and contribute back to the rostest docs.

PR #578 (2019-11-04): Fixed nondeterministic bug in joystick test by creating helper class
# Description
We've had a non-deterministic joystick test bug for a long time.

This PR does the following:
- Fixes the damn thing.

Fixes #575 

# Testing steps (If relevant)
## Run tests
1. Run `catkin_make run_tests_igvc_platform`
2. Repeat 1 as many times as you want

Expectation: It doesn't fail

To be honest, if there still was a race condition, you would probably need to run it like a few thousand times in order for it to show itself. I think the current code doesn't have a race condition though, so until it shows itse [... cut]
closed 2019-11-04 by oswinso