RoboJackets/igvc-software #588: Design Document on Firmware Rewrite

oswinso, 2019-11-08:
I'd like us to rewrite our firmware and communications code, on both the PC and the mbed side. Before we do that though, I think we should do a design document to properly design this out.

AC: Design document for firmware and communications between PC and MBED.

oswinso, 2019-11-08:
Just a few features I had in mind that I think would be nice to have

1. Proper use of RTOS. We currently have a coupled single-threaded controls and communications loops
2. Faster everything. I benchmarked communications, and was able to get like 1 MHz round-trip.
3. Move away from protobuf (heavy) to something like flatbuffers or even raw structs
4. Better communication of everything to PC
5. Better use of LEDs for status
6. Watchdog

Dallas-D, 2019-11-12:
Right now the logic board has 2 new status LEDs that can be added. We have the potential to have up to 5 status LEDs, but I think we need to have a discussion about what actually needs a status light and how useful they will be.

@oswinso For faster comms, what do you expect instead of 1kHz roundtrip? Also, what do you mean by better communications? Do you mean more efficient communications? Do you mean there's more information being sent from the mbed to the PC?

oswinso, 2019-11-12:
You should check out the firmware code located [here](https://github.com/RoboJackets/igvc-firmware/blob/master/src/mbed/main.cpp) for details, but
- My point with the round trip was to show that we have a lot of room for much faster communications between the PC and the mbed. We're currently only communicating at like 50 Hz, but I think we should do like 100 - 200 Hz
- By better communication, I mean like having better structured packets, as well as providing more information from the mbed to the computer. I think there are many pieces of information that we could send and should send from the mbed to the computer but aren't doing right now, such as shutdown cause, motor output values. Another point is of course, as you stated, more efficient communications. I think that some pieces of information are more essential (ie. encoder data) which should be sent back to the PC more frequently, while other pieces of data (ie. debugging, logs) should be sent back probably less frequently, or a configurable frequency, However, with how the communications are setup right now, we just have a big "response" message that _might_ contain everything, and I just think that it would be better to split this up into different packet types.

PR #626 (2020-01-12): Firmware Refactor Design Document
# Description

This PR is the design document for the firmware rewrite planned for spring 2020. The document itself can be read [here](https://github.com/RoboJackets/igvc-software/blob/design/firmware/documents/design/FirmwareRefactor.md).

There are the major changes proposed in the document:
- A new multi-threading scheme for improved efficiency
- Code structure changes for improved readability and maintainability
- New messages for communicating between the computer and the mbed
- More features and design problems to be aware of when implementing the system [... cut]
closed 2020-01-12 by Dallas-D