RoboJackets/igvc-software #618: Add style formatting / linting for CMakeLists.txt

oswinso, 2019-11-29:
The styling for our `CMakeLists.txt` is very varied and there isn't a standard right now, ie. parenthesis on newline or same line, how many spaces etc.

https://github.com/RoboJackets/igvc-software/blob/ac53f57ffd426bfbdb1252c44555f3ea77d9b276/igvc_navigation/CMakeLists.txt#L9-L38

https://github.com/RoboJackets/igvc-software/blob/ac53f57ffd426bfbdb1252c44555f3ea77d9b276/igvc_navigation/src/mapper/CMakeLists.txt#L2-L10

https://github.com/RoboJackets/igvc-software/blob/ac53f57ffd426bfbdb1252c44555f3ea77d9b276/igvc_platform/CMakeLists.txt#L106-L110

https://github.com/RoboJackets/igvc-software/blob/ac53f57ffd426bfbdb1252c44555f3ea77d9b276/igvc_platform/CMakeLists.txt#L150-L153

We should use a tool / linter / styler to make sure all our `CMakeLists.txt` are styled the same way.

matthewhannay567, 2020-02-20:
Found a library that does this, don't know how to implement it though

https://github.com/richq/cmake-lint

oswinso, 2020-02-20:
That library doesn't seem to be maintained though... :(