RoboJackets/igvc-software #651: Actual Barrel Detection With Neural Network

FZ2000, 2020-02-16:
A more robust solution compared to the Clustering and RANSAC solution. Basically treat LIDAR data like images and feed it to a neural network. A related paper could be found here: https://arxiv.org/abs/1612.00593

suhacker1, 2020-05-09:
@oswinso Should this be closed now or no?

oswinso, 2020-05-09:
This is barrel detection using LiDAR, which is different.

yyajima21, 2020-06-21:
@suhacker1 @oswinso I took over this project last week. Here is a to do list for completing our point cloud based segmentation model. I trained a semantic segmentation model using Stanford building datasets. Here is the repository I used for training the point cloud dataset.
link: [https://github.com/yanx27/Pointnet_Pointnet2_pytorch](url)

**To do list:**
- [ ] Search existing annotation tool to label point cloud data
- [ ] Develop a tool to save point cloud data from existing bag files.
- [ ] Modify a data loader code to accept our input data
- [ ] Add a tensor board feature