RoboJackets/igvc-software #721: Create model for swerve robot in gazebo.

matthewhannay567, 2020-05-11:
To start testing swerve bot code, we will need a model of a swerve bot in gazebo

JasonGibson274, 2020-08-18:
- [ ] Create a an example serve module. Specifically a cylinder leg and use the Wheel.dae scaled to the correct size. This will be an a separate xarco file.
- [ ] Write a ROS node to control the example swerve module. Speed set point for spinning the wheel and a position set point for rotating the wheel.
- [ ] Then create 4 of them and make a node that controls all the wheel at the same time in a separate xacro file called swervi.urdf.xacro . Do not worry about the rest of the robot. I want to see 4 peg legs and some floating wheels.
- [ ] Integrating all the sensors back into the new module.

Vikram-T, 2020-09-13:
Here's the rough dimensions for Swervi:
[Swervi Extremely Tentative Dimensions.pdf](https://github.com/RoboJackets/igvc-software/files/5215088/Swervi.Extremely.Tentative.Dimensions.pdf)

Vikram-T, 2020-09-13:
And documentation wise:
https://github.com/RoboJackets/igvc-software/blob/master/igvc_description/urdf/jessii.urdf.xacro is useful for reference

From what I figured it seemed that we need to create the robot in urdf in order to have it work with ROS. So I was using http://wiki.ros.org/urdf/Tutorials 
http://gazebosim.org/tutorials?tut=ros_urdf&cat=connect_ros to figure that out. 
* `<visual>` and fixed `<joint>`: [link](http://wiki.ros.org/urdf/Tutorials/Building%20a%20Visual%20Robot%20Model%20with%20URDF%20from%20Scratch)
* Moveable joints : [link](http://wiki.ros.org/urdf/Tutorials/Building%20a%20Movable%20Robot%20Model%20with%20URDF)
* `<inertial>` (Required in order to spawn in gazebo) : [link](http://wiki.ros.org/urdf/Tutorials/Adding%20Physical%20and%20Collision%20Properties%20to%20a%20URDF%20Model)
* http://gazebosim.org/tutorials/?tut=ros_control Is useful for understanding ros control which controls the joints and uses `<transmission>` tags

Vikram-T, 2020-09-13:
Right now I basically have a single wheel and a cylinder leg that is spawned in gazebo on the branch `swerve-xacro-model`. But it needs to be modified:
* It needs 4 links: 
   1. a base link (center of robot), 
   2. leg
   3. wheel hub
   4. wheel

You would connect the wheel hub to the wheel to make the wheel spin and the leg to the wheel hub to spin the whole assembly.
You should have two transmissions total on that platform. static link from base_link to the leg.

PR #819 (2021-01-07): Feat/swerve sim
# Description
This PR adds a Swervi robot simulation. The robot consists of 4 submodules, each with its own swivel and wheel. The robot also has a Lidar sensor but is missing cameras, IMUs, and the GPS sensor, as the specific position of these sensors is yet to be determined by the mechanical subteam. Those can be easily added later.

This PR does the following:
- Adds a Swervi model that can be added to future simulations.

Fixes #721 

# Testing steps (If relevant)
1. run `roslaunch igvc_description simulationTesting.launch`
2. observe the robot in Gazebo and make sure all th [... cut]
closed 2021-01-07 by andrewyarovoi