RoboJackets/igvc-software #777: Add an additional output topic for pointcloud_filter that removes the ground points.

matthewhannay567, 2020-10-07:
AC: pointcloud_filter publishes to an additional topic which filters out the ground plane points.

yyajima21, 2020-11-02:
I #806 did we ever check the output of point cloud filter after noetic migration? I don't think the noetic branch has a working point cloud filter. This is the error message I got.

[image]

yyajima21, 2020-11-13:
Update: I found out that our point cloud filter folder is outdated. So, I will make another package that removes a ground plane.

matthewhannay567, 2020-11-13:
What do you mean "outdated"? And when you say "make another package", do you mean another filter .cpp file?

PR #818 (2021-01-07): Feat/pointcloud clustering
# Description
This project contains a software package that segments lidar point cloud into clusters and creates a bounding box over each cluster in real time. This package adds a feature to localize objects surrounded by the robot such as barrels and people. Here is [readme file](https://github.com/RoboJackets/igvc-software/tree/feat/pointcloud_clustering/igvc_perception/src/pointcloud_segmentation) for this project.

This PR does the following:
- Ground plane segmentation
- Point cloud clustering that categorizes lidar points into several clusters using two well kn [... cut]
closed 2021-01-07 by andrewyarovoi