RoboJackets/igvc-software #828: Retrain the neural network for parking lot navigation.

matthewhannay567, 2021-02-28:
Because this year's IGVC competition will be on a parking lot, we need to train a neural network to work with a parking lot.

yyajima21, 2021-03-07:
@matthewhannay567, here are [some dataset](https://cloud.robojackets.org/apps/files/?dir=/RoboNav/Software/rosbags/2018-2019/parking-lot-2019&fileid=120997) collected from the parking lot. You may need to search some existing annotated dataset [here](https://cloud.robojackets.org/apps/files/?dir=/RoboNav/Software/dataset&). If you like to label some data, I used the following [website](https://github.com/wkentaro/labelme) to label images

VAM7686, 2021-07-19:
The trained binary segmentation parking lot model can be found on the Robojackets cloud under `RoboNav/software/model/IGVCModel_10_parkinglot.pt`

closed 2021-07-19 by VAM7686